ps2/rileylink_ios #2: Send ACK for all `a2` packets

ps2, 2015-08-18:
_From @loudnate on August 17, 2015 5:52_

Should we keep track of paired MySentry IDs, then?

_Copied from original issue: ps2/rileylink#23_

ps2, 2015-08-18:
_From @loudnate on August 17, 2015 6:6_

Pete suggested we don't have to worry about "real" MySentry devices, and of course we'd only ACK for the Pump IDs we know about.

loudnate, 2015-08-20:
@ps2 envisions a [stricter hierarchy](http://interactive.blockdiag.com/?compression=deflate&src=eJxLyslPzk7JTExXqOZSUEjOzyspys9R0LVTyM3My8xNTYnPTS0uTkxPRRYqSgPxknJSrblqAdNCFb4) for "`RLKit`". The control layer can accept a pump ID at initialization and automatically ACK all matching packets. Clients (e.g. NightScoutUploader) can subscribe to the control layer for notifications.

loudnate, 2015-09-06:
@ps2 Can you double-check what this MySentry response message is?

The one I send during pairing is:

`a2 PPPPPP 06 ++ MMMMMM 00 AAAAAAAA CC`
P = Pump ID
++ = Counter
M = MySentry ID
A = message types to acknowledge (up to 4, pad with 0)

If this is the proper message, the app would have to know (or better yet, derive) its MySentry ID.

ps2, 2015-09-14:
Yes, I believe to do a proper ack, one would need to remember the MySentry ID used to pair.  Does that answer your question?

loudnate, 2015-09-15:
Yeah, I just wanted to confirm. I've been sending those ACK packets for ~2 weeks now (on channel 3) and the pump still sends 2 additional copies to follow.

loudnate, 2016-03-05:
Per my last comment, this doesn't seem to make a difference.

closed 2016-03-05 by loudnate