ps2/rileylink_ios #10: Start using gitflow and semver release tagging

loudnate, 2015-08-22:
Sorry to feel so corporate, but as more folks start to use the app, some structure will help master be unbroken :innocent:

ps2, 2015-08-23:
No apologies needed.  It's a good idea.  Need to version the firmware as well.

loudnate, 2015-08-23:
In other projects, I bump `CFBundleVersion` each time an archive operation is run using a script. I'd we could bump it on each build, but that will cause unnecessary merge conflicts since we're not using a centralized build server. For now I think we'll increment each of `CFBundleVersion`, `CFBundleShortVersionString`, and `RileyLink.podspec` by hand after each `dev` branch merge.

If you're up for writing a ruby command-line script to keep all three in sync, that'd be great (I am quite rusty in ruby myself), and I can configure it as part of the build process.

ps2, 2015-08-23:
Were you thinking the ruby script would just verify that things were in sync, or actually take the RileyLink-Info.plist key and update the podspec?  Or look at the podspec and update the plist?

ps2, 2015-08-23:
Notes: plutil can update plists, and we can turn the podspec into json with pod ipc spec

loudnate, 2015-08-23:
It's probably easier to make Info.plist canon, specifically CFBundleShortVersionString since it has a friendly UI in the project inspector.

commit 598100a8 (2015-08-25): Bumping version to 0.1.1

Fixes #10
PR #15 (2015-08-31): Check version before each build
@ps2 This will simply halt the build with the error printed by the script. 
Because this is running as a pre-build phase, it's legal to modify source files here if we wanted to change the script to write the version numbers out to each location, choosing one as a source-of-truth.

Fixes #10
PR #14 (2015-09-03): Version 0.1.1
Leave this open until we're ready to release. (Fixes #10)

* Improves the user experience of the MySentry Pairing screen
* Fixes layout issues on the packet list screen on larger iPhones
* Removes unused battery service
* Includes bulid script to keep version strings in sync
* Working towards CocoaPods support (podspec is still a draft)
closed 2015-08-31 by ps2