ps2/rileylink_ios #22: Ensure central manager is powered on before subscribing to characteristics

loudnate, 2015-09-05:
When the app is re-launched in the background, the central manager's state is restored with "connected" peripherals, before the CBCentralManagerState is powered on. Writing, reading, or subscribing to characteristics during that time results in the following warning:

```
Sep  4 18:36:04 Nates-iPhone-6 Naterade[4360] <Warning>: (Warn ) CoreBluetooth: API MISUSE: <CBCentralManager: 0x170269000> can only accept this command while in the powered on state
```

ps2, 2016-06-23:
I believe this is resolved, no?

loudnate, 2016-06-23:
No, I still haven't gotten around to it. Other than a console error of "API
MISUSE" on app startup, there's no other side effects (at this time).

ps2, 2017-11-09:
Fixed in iOS 11 patch.

closed 2017-11-09 by ps2