ps2/rileylink_ios #24: Interface to select which meter ids to accept mbg messages from

ps2, 2015-09-16:
Would be nice if we could get this info from the pump!  Is there a way to dump paired meter ids?

loudnate, 2015-09-16:
Maybe?
https://github.com/bewest/decoding-carelink/blob/master/decocare/commands.py#L1363

loudnate, 2015-09-16:
### PUMP MODEL: `ReadPumpModel:size[64]:data:'723'`
###### sending `<class 'decocare.commands.ReadOtherDevicesIDS'>`
response: ReadOtherDevicesIDS:size[64]:data:bytearray(b'\x01\x00\xa2\x00\x08\x88\x00\x00\x00\xa5S\x11O\x0f\x00\x16\x01\xa5S\x11O\x0f3<\xb5X\x11O\x0f\x00\x16\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
#### decoded:
```python
bytearray(b'\x01\x00\xa2\x00\x08\x88\x00\x00\x00\xa5S\x11O\x0f\x00\x16\x01\xa5S\x11O\x0f3<\xb5X\x11O\x0f\x00\x16\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
```

loudnate, 2015-09-16:
Meter: 783436
"MySentry": a2000888

loudnate, 2015-09-16:
I see mysterious "a5" entries but I don't see that meter ID.

loudnate, 2015-09-16:
The meter ID does appear in the heading of the carelink CSV, so there should be a command somewhere: https://github.com/bewest/decoding-carelink/blob/5f224e0b0019e4b9ad9788dcd10d9281468ccfdf/analysis/736868/cgm/CareLink-Export-1428895306376.csv