ps2/rileylink_ios #25: Avoid unecessary handling for duplicate packets.

ps2, 2015-09-16:
A simple "is this the same packet as the last one we handled, and if so, drop it" should cut down on most of the duplicates.

loudnate, 2015-09-16:
Yeah, [that's what I'm doing](https://github.com/loudnate/naterade-ios/blob/master/Naterade/Managers/PumpDataManager.swift#L98-L100), though equality is defined as `lhs.pumpDate == rhs.pumpDate && lhs.glucoseDate == rhs.glucoseDate` since the sequence numbers will differ.

ps2, 2016-08-07:
Same logic is done in both Loop and RileyLink.  Though I would like to record what the actual MySentry does, as I remember it had a way of ACKing that avoided subsequent packets being sent from the pump.

loudnate, 2016-08-08:
The original problem with my attempts to ACK was not replying with the sequence number. You figured that out during refactoring of the sentry pairing, so maybe we should try again. 

I'd love to get that 10s back every 5 minutes to ease my phone's battery usage.