ps2/rileylink_ios #37: Poll Pump History

ps2, 2015-11-11:
Poll pump history, and extract treatments, meter bg, etc.

loudnate, 2015-11-11:
I've been thinking a lot about this, and I'm not sure there's enough background quota to manage the pump power wake and download history. What do you think about moving power control to the firmware, with a write/notify characteristic when it's on?

ps2, 2015-11-11:
I think that's a good idea from a system design perspective (and maybe necessary, as you point out).  The two things I don't like about it are: 1. we have to write it in bgscript, and 2. lots of users don't have the ability to upgrade firmware. But those are perhaps just unpleasant realities we need to accept. :)

loudnate, 2015-11-11:
Yeah, and maybe I'm putting the cart before the horse. 

In between reverse-engineering the G5, I have started to work on a process for incremental updates of pump history, marking records as final or mutable. Each page of history we can avoid will save a lot of time (and potential error). 

For someone like me who is auto-issuing temp basals, the history pages fill really fast, so this will be a benefit even on a carelink setup.

ps2, 2015-11-21:
Which records in pump history are mutable?

loudnate, 2015-11-21:
Bolus records (normal and square) are, though bolus cancellation is sufficiently rare. 

TempBasal records aren't mutable but can be superseded.

ps2, 2016-01-06:
How would one mutate a bolus record after the fact?

loudnate, 2016-01-06:
bolus records (square and normal) keep track of the amount delivered, and and update it during the dose.

ps2, 2016-07-19:
I think we have the methods called for at this layer.  History reconciliation is another layer, I think.

PR #39 (2015-12-23): Pump history
Currently polling page 0, and decoding into types; specific fields are not being decoded yet.

Fixes #37
closed 2016-07-19 by ps2