ps2/rileylink_ios #42: MessageSendOperation main vs start

ps2, 2015-12-28:
Review moving the code in MessageSendOperation main() back to start(). I had moved it when I encountered issues cancelling operations before they started, but that might not be the right fix.

ps2, 2015-12-29:
Maybe related:

```
2015-12-28 21:22:01.612 RileyLink[1180:481314] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSOperationInternal _start:]: something is trying to start the receiver simultaneously from more than one thread'
*** First throw call stack:
(0x1840b9900 0x183727f80 0x1840b9848 0x1849a320c 0x184a62728 0x1001edbb0 0x1001f3658 0x184070bb0 0x18406ea18 0x183f9d680 0x1854ac088 0x188e14d90 0x1001033b4 0x183b3e8b8)
libc++abi.dylib: terminating with uncaught exception of type NSException
```

loudnate, 2016-03-05:
Won't Fix!

closed 2016-03-05 by loudnate