ps2/rileylink_ios #48: incorrect bolus reporting to NS.

ps2, 2016-01-22:
```5b0020540974100f5000784b500000300000000030965c05684bc0010030003000400020544974107b0400400a1410140b000a7a30752b74103f0f30754b7410c527ad5b7a067a0b7410265000b44b500000540000040054965c0830a3c068e9c00100540054000400067a4b74107b0500400c1410180a005b00356b0f74100f5000b44b500000200000000020965c0b54e4c03084d068cad00100200020000000356b4f74107b060040101410200e007b07004013141026100021002f6e131410030000003c1a6f3314107b0703401414102610000300030003347b1314100a3e25503414905b3e2750141410005100784b505800000000000058965c052019d0010058005800000027505414105b3e20581474100c5100784b505800280000580028965c085809c02021d0010028002800580020585474105b000a5d1474100f5000784b500000300000000030965c0b0c04c0740ec02026d001003000300080000a5d5474107b0800401514102a13000a4f0c5a3674103f090c5af67410c527ad0a5a005d3774103f0b005d577410c527ad331e095f171410081601095f17141007000002f014900000006e149005408e4f7a05000002f001342901bc3b007f0164005800000000060100000400000000000000003e3e00000000000000007b000941001510000e007b0100400115100208007b020040041510080c007b0300400615100c10000a5a0e722775103f0b0e72477510c527ad5b003577087510125000784b5000003c000000003c9601003c003c00000035774875107b0400400a1510140b000ae130722b75103f1c30722b7510c527ad5be1017a0b75103d5000b44b5028008400000400a8965c053cb7c05be1017b0b75103e5000b44b5028008800000000b0965c053cb8c05be1167b0b75103e5000b44b5028008800000000b0965c053cb8c07b0500400c1510180a000100880088000000167b4b75107b060040101510200e005b003a411075100f5000784b500000300000000030965c0888f6c03caad001003000300000003a415075101e010b6d1015107b07055e1315102610001f20055e1315105b002866131510295000784b500000880000000088965c0830dfc088cfd0010088008800000028665315107b0800401515102a13000abd00583575103f170058b57510c527ad5bbd05591515101e5000b4555010004000002c0040965c08886ec0304ad000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000ec33```

ps2, 2016-01-22:
The above history page ends with a BolusWizardBolusEstimate and no BolusNormal.  This is unusual, I think.  The history fetch happened right around the same time that I gave the bolus.  The code in NightScoutBolus.m produced bolus entries with the wrong number of carbs attached to them.  It seemed like the carbs from a subsequent (or prior?) bolus were attached.

ps2, 2016-01-22:
Since the code that does this is based on https://github.com/openaps/oref0/blob/master/lib/bolus.js, should check that code as well for this error.

loudnate, 2016-01-22:
https://github.com/loudnate/openaps-mmhistorytools/blob/master/openapscontrib/mmhistorytools/historytools.py#L70-L76

loudnate, 2016-02-09:
https://github.com/tidepool-org/chrome-uploader/blob/master/lib/carelink/bolusNWizard.js

loudnate, 2016-02-09:
Tidepool has handled this issue pretty well. Note that they break out X22/X23 parsing separately

ps2, 2016-06-23:
The new code in NightscoutPumpEvents.swift handles this correctly, I believe.

closed 2016-06-23 by ps2