ps2/rileylink_ios #74: Assert PumpModel string in PumpState for more comms

loudnate, 2016-04-19:
And manage a signal to trigger serializing PumpState after model/timeZone changes.

loudnate, 2016-04-20:
Thinking through how best to notify observers of PumpState changes today.
KVO is the right paradigm, but it means quite a bit of boilerplate for observers, and it feels so archaic in Swift. It also requires the observer to extend `NSObject`, which I'd like to avoid.
Beyond that, it's really either rolling our own interface:
```swift
addObserver((keyName: String, oldValue: Any?, newValue: Any?) -> Void) -> ObserverID
removeObserver(ObserverID)
```
Or using `NSNotificationCenter`, which is a little more heavyweight but doesn't add a custom variation on a concept.

ps2, 2016-04-20:
I like the idea of KVO, but find it unwieldy.  And we're already using NSNotifications everywhere, right?

loudnate, 2016-04-20:
Yeah, we are, but I'm looking into move a few of those back to simple delegation where 1-M relationships aren't necessary. I believe `PumpState` has the need for multiple observers, however, so I think I'll start there.

PR #103 (2016-06-03): Actively populate PumpState.pumpModel when known
Fixes #74
commit cb8de2c8 (2016-06-03): Actively populate PumpState.pumpModel when known

Fixes #74
closed 2016-05-23 by loudnate