ps2/rileylink_ios #140: UnknownEventType(70)

ps2, 2016-06-28:
```
2016-06-28 17:59:54.824 RileyLink[1728:714632] Fetched page 0: <6e791005 00616161 01000001 57015764 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 7b014080 011a1002 0c007b02 4080041a 10080d00 7b034080 061a100c 10007b04 40800a1a 10140b00 7b054080 0c1a1018 0a000a74 64832c7a 103f0e64 838c7a10 c527ad7b 06408010 1a10200e 007b0740 80131a10 2610007b 08409e14 1a102915 007b0040 80001b10 00100007 00000157 7a100000 006e7a10 05007474 74010000 01570157 64000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 007b0140 80011b10 020c000a 8d549a21 7b103f11 549aa17b 10c527ad 7b024080 041b1008 0d007b03 4080061b 100c1000 7b044080 0a1b1014 0b007b05 40800c1b 10180a00 7b064080 101b1020 0e007b07 4080131b 10261000 7b08409e 141b1029 15007b00 4080001c 10001000 07000001 577b1000 00006e7b 1005008d 8d8d0100 00015701 57640000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00007b01 4080011c 10020c00 7b024080 041c1008 0d007b03 4080061c 100c1000 7b044080 0a1c1014 0b007b05 40800c1c 10180a00 500059ba 0f1c1020 011e00b4 14001e3c 24ef6db4 4600803c 21011e00 b414001e 3c24ef6d b4460080 3c0b6b00 40bb2fbc 107b0640 80101c10 200e0050 00468110 1c102101 1e00b414 001e3c24 ef6db446 00803c21 011e00b4 14001e3c 25c123b4 4600803c 5b007ba9 111c100a 5000784b 50000021 00000000 21960100 21002100 00007ba9 511c1000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0000546e>
2016-06-28 17:59:54.830 RileyLink[1728:714632] Finished running dispatched RL comms task
2016-06-28 17:59:54.830 RileyLink[1728:710186] History fetch failed: UnknownEventType(70)
```

loudnate, 2016-06-28:
Do you know the length? 

Go ahead and add the opcode to the enum, it will now try to use UnknownPumpEvent for parsing.

ps2, 2016-06-28:
I think I figured it out; incorrectly parsing the previous entry...

PR #142 (2016-07-02): Fix ChangeSensorSetup2PumpEvent parsing on 551 pumps
Fixes #140
closed 2016-07-02 by loudnate