ps2/rileylink_ios #147: Add a method to test/verify Nightscout credentials

loudnate, 2016-07-04:
Can we add a method that does a no-op call to the Nightscout API using the provided credentials, to help users validate that it's working immediately after entering them?

@ps2 as someone who still hasn't setup Nightscout, I'm looking to you for guidance. The API endpoints I've seen seem to be read-only public; is that true for everybody? I'm hoping there's a way test both the URL and the secret.

ps2, 2016-07-10:
@mhaeberli mentioned an endpoint specifically for testing auth. Something like ` https://xxxsite.azurewebsites.net/api/v1/experiments/test `  

Just adding as a note for when I look into this.

loudnate, 2016-07-10:
https://github.com/nightscout/cgm-remote-monitor/wiki/API-v1.0.0-beta-Security

closed 2016-07-21 by loudnate