ps2/rileylink_ios #200: Ability to use MM CGM with an x22 Pump

tomasboudr, 2016-08-17:
Looking for the ability to use a MM CGM with an x22 pump, thus we will need to parse sensor data pages in rileylink

Related issue in Loop: https://github.com/loudnate/Loop/issues/100

ps2, 2016-09-27:
See getHistoryEvents in https://github.com/ps2/rileylink_ios/blob/dev/RileyLinkKit/PumpOpsSynchronous.swift for an example of the swift code you'll need to write to fetch the pages

ps2, 2016-09-27:
@bewest  has Python code to decode CGM pages here: https://github.com/openaps/decocare/blob/master/decocare/cgm/__init__.py

tmecklem, 2016-10-15:
I'm spiking a solution to this at https://github.com/tmecklem/rileylink_ios/tree/support-x22-cgm.

ps2, 2016-11-01:
Support for fetching CGM pages has been merged in.  Still not uploading CGM data to NS.

ps2, 2016-12-02:
Fixed with #275

closed 2016-12-02 by ps2