ps2/rileylink_ios #205: Upload suspend and unsuspend events to NS

ps2, 2016-08-21:
These events are not yet handled in NS, but they are important for remote monitoring.

loudnate, 2016-08-21:
I assume the goal is to upload as-raw-as-possible pump data, and any analysis/reconciliation of that pump data into actual insulin delivery would happen in Nightscout?

ps2, 2016-08-21:
Yes, I think so. I would like to make it work without too much difficulty with the temp basal display. I think storing it as another treatments.json record makes sense and would work with the existing code in NS well.

loudnate, 2016-08-21:
gotcha. I've got a slew of unit test fixtures for properly reconciling that data if you need it.

ps2, 2016-08-21:
Is that https://github.com/loudnate/openaps-mmhistorytools ?

loudnate, 2016-08-21:
And https://github.com/loudnate/LoopKit/tree/master/InsulinKitTests

ps2, 2017-03-09:
Fixed with #306

closed 2017-03-09 by ps2