ps2/rileylink_ios #218: incorrect maxBolus in  ReadSettings response on x22 pumps

ecc1, 2016-09-03:
While debugging my own code and comparing it to yours, I think I found a bug here:
https://github.com/ps2/rileylink_ios/blob/dev/MinimedKit/Messages/ReadSettingsCarelinkMessageBody.swift#L56

I may be misunderstanding your Swift code, but on a x22 model pump, the maxBolus value is in byte 6, not byte 7.
See https://github.com/bewest/decoding-carelink/blob/master/decocare/commands.py#L1274
versus https://github.com/bewest/decoding-carelink/blob/master/decocare/commands.py#L1310
(Ben's indices are all off by 1 compared to yours (and mine)).

Here is some test data:

model 522 pump

C0 15 0C 02 00 05 01 78 00 64 00 00 01 00 01 00 00 64 01 03 00 0F

should produce (among other values):
```
{
  "AutoOff": "12h0m0s",
  "InsulinAction": "3h0m0s",
  "InsulinConcentration": 100,
  "MaxBolus": 12,
  "MaxBasal": 2.5,
  "RfEnabled": true,
  "TempBasalType": "Absolute",
  "SelectedPattern": 0
}
```
model 523 pump

C0 19 0C 02 00 01 01 00 8C 00 50 00 00 00 00 00 00 64 01 03 00 14 00 64 00 00

```
{
  "AutoOff": "12h0m0s",
  "InsulinAction": "3h0m0s",
  "InsulinConcentration": 100,
  "MaxBolus": 14,
  "MaxBasal": 2,
  "RfEnabled": false,
  "TempBasalType": "Absolute",
  "SelectedPattern": 0
}
```

ps2, 2016-09-23:
What is the RFEnabled field?

ecc1, 2016-09-23:
On Thu, Sep 22, 2016 at 07:02:38PM -0700, Pete Schwamb wrote:
> What is the RFEnabled field?

I took this as-is from decocare. I think it's true when either a
remote or a meter has been enabled, otherwise false (but I haven't
verified that).

--
Eric Cooper             e c c @ c m u . e d u

ps2, 2016-09-23:
I just tested on a 522, with some random max bolus values, including 24.9, and this is parsing it correctly.

ps2, 2016-09-23:
Here's the response from my 522 when I set the max bolus to 24.9:
```
a7 350535 c0 19000400010000f90078000000010000640103000a0032010101000000000000000000000000000000000000000000000000000000000000000000000000000000b0
```

ps2, 2016-09-23:
I just tested on a 551 (530G) and it's parsing correctly to.  Perhaps this is a 5 vs 7 issue?.

ps2, 2016-09-23:
Just tested on a 723 and it's working.  Going to close.  If you think I'm still missing something, let me know, and I'll reopen it.

ps2, 2016-09-23:
Ah, grabbed the wrong pump; thought I had a 522, but it was a 522.. Will retest on the 522.

PR #245 (2016-09-24): Fix ReadPumpSettings for x22 and older pumps.
Fixes #218
closed 2016-09-24 by ps2