ps2/rileylink_ios #243: getHistoryEvents(since: should go back farther for x22 pumps

loudnate, 2016-09-22:
[image]

Here's an example of what we recently discovered: bolus events are written after completion, and potentially out-of-order, for x22 pumps.

We still need to change [this condition](https://github.com/ps2/rileylink_ios/blob/dev/RileyLinkKit/PumpOpsSynchronous.swift#L483), I believe.

ps2, 2016-09-23:
I think you're right about that condition changing.  If we change it, though, we'll be returning the same items multiple times from this call; we may need to add some de-duping code.

loudnate, 2016-09-23:
We already have to dedupe a small number of records on every call, because we use >= the last record date. 

LoopKit is using a unique constraint on (date, raw data), which handles the current cases (I believe).

loudnate, 2016-09-23:
If we need to minimize the repetition, we could consider a property on the PumpEvent protocol called "canBeOutOfOrder" and only return those

ps2, 2016-09-23:
I start to wonder if tracking by page # and record index is a better idea than time.  If we could track that, we'd only have to track as far back as the last modifiable record (which only exist on x23 and newer pumps).

loudnate, 2016-09-23:
Isn't the most recent page always page 0? CGM pages truly ring, but I don't think history does (externally).

ps2, 2016-09-23:
yes, but I think there is a getPageNumber command that returns a counter that you can use to detect rollovers.  We could also generate a page hash with the first record in the page.

ps2, 2016-09-23:
I think this might suffice: f19575a

loudnate, 2016-09-23:
> yes, but I think there is a getPageNumber command that returns a counter that you can use to detect rollovers. We could also generate a page hash with the first record in the page.
excellent – you're sure that `getPageNumber` isn't returning the glucose page number, right?

closed 2016-09-27 by loudnate