ps2/rileylink_ios #284: Loop RileyLink functionality freezing issue

ruess, 2016-12-02:
Moved from Loop Issue tracker to rileylink_ios issue tracker:
==============
Hey all,

Spoke to a few folks on Gitter and confirmed that it's happening to a few other people as well. Basically when Loop temporarily loses connectivity to RileyLink for a bit (guessing 15-20 mins), it will permanently stay disconnected, even if both are near each other and should be working. The only way to resolve the issue is a restart of the Loop app (note: a restart of RileyLink is NOT required).

You know you need to restart Loop because when you go into the Settings->RileyLink and try pressing on any of it's functions such as tuning the radio, getting pump model, or otherwise, absolutely nothing happens. It's completely frozen. To be clear, it's not Loop as a whole that's frozen as most things outside of the RileyLink screen still work, but it's the RL screen itself that's mostly frozen. Here's a video of me trying a number of RileyLink/pump functions in Loop after the freeze and as you can see nothing happens:
https://www.youtube.com/watch?v=oRMiyjWGvpc
(note that you can't actually see the "cursor" which is my finger depressing these functions, but I am and as you can see nothing happens)
-Also here's a gist of the crash report for when this happened grabbed from my iPhone: https://gist.github.com/ruess/dbad012d1f0dae85d6b754df5debdfdf

As noted above, after a simple restart of the app, things work again - however it's important to stress that it's not RileyLink which needs to be restarted, it's Loop.

While this isn't mission critical, it's happened once where things got disconnected somehow during the night and I awoke high because the 2 (Loop + RL) never connected again once they disconnected.

I'd like to see if there's a way to build in some sort of RileyLInk connection checker which gets triggered after more than 10 minutes of Red status. The way it would work is that if RileyLink wasn't detected, it would try restarting JUST the Loop functionality service that communicates with RL within the app in the hopes that it could potentially reconnect again without requiring the user to restart the app.

Any ideas where to start here? Basically having not dived into the code (and not being a huge coder myself), what kind of services within the app would be good candidates to try restarting (toggling on/off) in terms of RileyLink communication? Also, any ideas of where to place this feature? I'm guessing there may already be some activities that automatically take place when it turns to Red status in order to see what it can do to reconnect. Can someone point me to that stuff?

Let me know your thoughts and any ideas you might have about it.

=======================
 @ps2
 Collaborator
ps2 commented 2 days ago
Thanks, @ruess. This is an issue that's been around quite a while. Some people see it more frequently than others, it seems. I've only seen it rarely, and when the RL has been away from the phone for some time.

The more common diagnostic test is to look at the connection status. It will say 'Connecting...'.

One possible attempt to solve it that we've talked about is to see if resetting the core bluetooth central manager would fix it.

To test that theory, I would propose we do it in two steps. First we could make a version of the RL frameworks that has a new command to reset the central manager. Then user can manually try the command when they encounter the situation, and see if the problem resolves. If that does work, the second step would be to try to attempt to automatically detect the problem and automatically reset the central manager.

Here is where the central manager is initialized: https://github.com/ps2/rileylink_ios/blob/dev/RileyLinkBLEKit/RileyLinkBLEManager.m#L54
@ps2
 Collaborator
ps2 commented 7 hours ago
This is also probably better tracked as an issue in the rileylink_ios repo
@ruess
     
=============================
ruess commented
Hey Pete,

Thanks for the detailed explanation and guidance! I think starting out with a button or switch of some sort that the user can try to restart things when this happens is a great idea.

Also, I'll copy over everything into the rileylink_ios repo and close this one out.

Thanks!

shanselman, 2017-04-11:
FWIW I get this every day. If there is some guard code or a way I can get more info, let me know. Resetting BT or force quitting gets me going.

ps2, 2018-01-24:
This may have been fixed in v1.2.0