ps2/rileylink_ios #300: Crashes from `Unbalanced call to dispatch_group_leave()`

tmecklem, 2017-01-06:
There are two recurring stacktraces in crash logs related to `Unbalanced call to dispatch_group_leave()`, both in RileyLinkBLEDevice. 

The stacktraces seem to indicate that the lines of code being executed at the time of the crashes are:

* https://github.com/ps2/rileylink_ios/blob/dev/RileyLinkBLEKit/RileyLinkBLEDevice.m#L147
* https://github.com/ps2/rileylink_ios/blob/dev/RileyLinkBLEKit/RileyLinkBLEDevice.m#L426

I'll attach the relevant stacktraces below.

tmecklem, 2017-01-06:
Crash log from `- (void)dataReceivedFromRL:(NSData*) data`

[com.mecklem.loop_issue_2_crash_bf816d6c0e164ff09798a0aa8aeca723_c3299731043345c994f545fb675a9ca1_0_v1.txt](https://github.com/ps2/rileylink_ios/files/688949/com.mecklem.loop_issue_2_crash_bf816d6c0e164ff09798a0aa8aeca723_c3299731043345c994f545fb675a9ca1_0_v1.txt)

tmecklem, 2017-01-06:
Crash log from `- (BOOL) doCmd:(nonnull CmdBase*)cmd withTimeoutMs:(NSInteger)timeoutMS`

[com.mecklem.loop_issue_3_crash_310f6efe13bd47aeadd2d6e53b26807d_4d20652ff56546db95cd412f0e8f4591_0_v1.txt](https://github.com/ps2/rileylink_ios/files/688953/com.mecklem.loop_issue_3_crash_310f6efe13bd47aeadd2d6e53b26807d_4d20652ff56546db95cd412f0e8f4591_0_v1.txt)

tmecklem, 2017-01-06:
And finally a desymbolicated crash: 
[com.mecklem.loop_issue_1_crash_6b6608ca932745b9a5aa6b172419140c_9e366a0174774ee6b9dc0887324d800e_0_v1.txt](https://github.com/ps2/rileylink_ios/files/688968/com.mecklem.loop_issue_1_crash_6b6608ca932745b9a5aa6b172419140c_9e366a0174774ee6b9dc0887324d800e_0_v1.txt)

ps2, 2018-01-24:
This is fixed in dev.