ps2/rileylink_ios #303: Issue Setting Temp Basal on MiniMed 522

danyocom, 2017-01-19:
I have been playing around with my own IOS app using RileyLinkKit on my 522 pump. Everything I have tested works except for calling "setTempBasal" (in PumpOps.swift).  

Is there a technical limitation that makes the 522 unable to accept tempBasal remotely?  

Any "tricks" to getting make the setTempBasal function in RileyLinkKit work on a 522 pump?

ps2, 2017-01-20:
522's do accept temp basal.  Make sure your temp basal units are on percent, and that the temp basal you're trying to set is under your max basal rate setting on your pump

danyocom, 2017-01-23:
I bumped the max basal rate up to as high as it goes, and have tried setting the temp basal type on the unit to both % and u/h.   The error message I am getting is rfCommsFailure("Could not verify TempBasal on attempt 2").  Any other suggestions? 

setting a Temp basal is the only functionality that I have not been able to get working.

danyocom, 2017-01-23:
After digging through the code some more, I realized I was trying to set a temp duration that was shorter that the minimum duration supported by my pump (30 mins). Everything is working now!

closed 2017-01-23 by danyocom