ps2/rileylink_ios #379: Nightscout treatments will be lost if they are the same type in the same second

trixing, 2018-02-03:
Nightscout deduplicates events by eventType and created_at in lib/server/treatments.js

NightscoutUploadKit uses TimeFormat which uses the ISO8601 formatter, which does the following:

formatter.dateFormat = "yyyy-MM-dd'T'HH:mm:ssX"

That way if we try to write 2 events in the same second the second one overwrites the first.

I'd suggest changing the format at least for the treatments to capture milliseconds

formatter.dateFormat = "yyyy-MM-dd'T'HH:mm:ss.SSSSX"

or use the solution from this stackoverflow hint to capture microseconds

https://stackoverflow.com/questions/43123944/how-to-configure-dateformatter-to-capture-microseconds/43124982

ps2, 2018-02-05:
I believe the pump clock only has time resolution of 1s.  If NS is dropping unique events that share the same timestamp, then this would not fix the issue.

trixing, 2018-02-05:
I noticed this behavior because I changed my AnalyticsManager and loggly library to log events to Nightscout instead of 3rd party resources (with a custom eventType).  Those sources will actually generate more than one event per second from time to time.