ps2/rileylink_ios #420: Unknown response in 523 pump setup

dm61, 2018-07-20:
In the latest Loop dev, which includes the new "Add Pump" setup. I am not able to go past the Pump Setup. In multiple retries, reboots of everything, the outcome is always the same:

`Unknown response during PumpMessage(carelink, selectBasalProfile, 3 bytes, 65 bytes): 23 bytes` 

so I suspect there is an issue in Pump Setup, possibly specific to my pump type, which is 523NAS with 2.3A firmware. Below are a screen shot and capture of the Xcode debug output in two consecutive attempts to connect. 
[image]
2018-07-20 12:44:43.934202-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 0102
2018-07-20 12:44:44.081767-0600 Loop[432:152292] [PeripheralManager+RileyLink] RL Recv(single): dd737562675f726673707920322e30
2018-07-20 12:44:44.083900-0600 Loop[432:152292] [PeripheralManager+RileyLink] RileyLink response: GetVersionResponse(code: RileyLinkBLEKit.ResponseCode.success, version: "subg_rfspy 2.0")
2018-07-20 12:44:59.605744-0600 Loop[432:151645] [MC] Lazy loading NSBundle MobileCoreServices.framework
2018-07-20 12:44:59.606836-0600 Loop[432:151645] [MC] Loaded MobileCoreServices.framework
2018-07-20 12:45:01.928760-0600 Loop[432:151645] [MC] System group container for systemgroup.com.apple.configurationprofiles path is /private/var/containers/Shared/SystemGroup/systemgroup.com.apple.configurationprofiles
2018-07-20 12:45:01.931165-0600 Loop[432:151645] [MC] Reading from public effective user settings.
2018-07-20 12:45:05.592313-0600 Loop[432:151645] -[UIWindow endDisablingInterfaceAutorotationAnimated:] called on <UITextEffectsWindow: 0x10610d400; frame = (0 0; 375 667); opaque = NO; autoresize = W+H; layer = <UIWindowLayer: 0x1c4235aa0>> without matching -beginDisablingInterfaceAutorotation. Ignoring.
2018-07-20 12:45:06.152509-0600 Loop[432:151645] Can't find keyplane that supports type 4 for keyboard iPhone-PortraitChoco-NumberPad; using 4072550144015629828_PortraitChoco_iPhone-Simple-Pad_Default
2018-07-20 12:45:06.169222-0600 Loop[432:151645] Can't find keyplane that supports type 4 for keyboard iPhone-PortraitChoco-NumberPad; using 4072550144015629828_PortraitChoco_iPhone-Simple-Pad_Default
2018-07-20 12:45:26.579395-0600 Loop[432:152291] [RileyLinkDevice] ======================== Pump ID Setup ===========================
2018-07-20 12:45:26.580210-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 010d
2018-07-20 12:45:26.709112-0600 Loop[432:152613] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:26.709615-0600 Loop[432:152613] [PeripheralManager+RileyLink] RileyLink response: CodeResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:26.710161-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060c99
2018-07-20 12:45:26.830208-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:26.830959-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:26.832429-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060f61
2018-07-20 12:45:26.952953-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:26.953637-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:26.961526-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 0306107e
2018-07-20 12:45:27.071516-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:27.071851-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:27.075725-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03061115
2018-07-20 12:45:27.193188-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:27.193745-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:27.199315-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060b99
2018-07-20 12:45:27.313547-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:27.314135-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:27.320823-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060a31
2018-07-20 12:45:27.433653-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:27.434062-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:27.434653-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060926
2018-07-20 12:45:27.552875-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:27.553401-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:27.561197-0600 Loop[432:152291] [PumpMessageSender] getResponse(PumpMessage(carelink, getPumpModel, 3 bytes, 1 bytes), 0, 0.180000, 1)
2018-07-20 12:45:27.563197-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 19050000000000000000b4010000a966726b18f468d555359000
2018-07-20 12:45:28.215447-0600 Loop[432:152613] [PeripheralManager+RileyLink] RL Recv(single): aa
2018-07-20 12:45:28.217787-0600 Loop[432:152613] [PeripheralManager+RileyLink] RileyLink response: PacketResponse(code: RileyLinkBLEKit.ResponseCode.rxTimeout, packet: nil)
2018-07-20 12:45:28.224931-0600 Loop[432:152291] [PumpMessageSender] getResponse(PumpMessage(carelink, powerOn, 3 bytes, 1 bytes), 255, 0.001000, 0)
2018-07-20 12:45:28.227930-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 190500ff00000000000001000000a966726b18f494d555356000
2018-07-20 12:45:32.739866-0600 Loop[432:152612] [PeripheralManager+RileyLink] RL Recv(single): aa
2018-07-20 12:45:32.740168-0600 Loop[432:152612] [PeripheralManager+RileyLink] RileyLink response: PacketResponse(code: RileyLinkBLEKit.ResponseCode.rxTimeout, packet: nil)
2018-07-20 12:45:32.741982-0600 Loop[432:152291] [PumpMessageSender] getResponse(PumpMessage(carelink, powerOn, 3 bytes, 1 bytes), 255, 12.000000, 0)
2018-07-20 12:45:32.742876-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 190500ff00000000002ee0000000a966726b18f494d555356000
2018-07-20 12:45:49.329561-0600 Loop[432:152687] [PeripheralManager+RileyLink] RL Recv(single): aa
2018-07-20 12:45:49.329870-0600 Loop[432:152687] [PeripheralManager+RileyLink] RileyLink response: PacketResponse(code: RileyLinkBLEKit.ResponseCode.rxTimeout, packet: nil)
2018-07-20 12:45:49.331330-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060b77
2018-07-20 12:45:49.451402-0600 Loop[432:152688] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:49.451822-0600 Loop[432:152688] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:49.454305-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060a2f
2018-07-20 12:45:49.577553-0600 Loop[432:152688] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:49.578008-0600 Loop[432:152688] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:49.578577-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 03060926
2018-07-20 12:45:49.691146-0600 Loop[432:152688] [PeripheralManager+RileyLink] RL Recv(single): dd
2018-07-20 12:45:49.691574-0600 Loop[432:152688] [PeripheralManager+RileyLink] RileyLink response: UpdateRegisterResponse(code: RileyLinkBLEKit.ResponseCode.success)
2018-07-20 12:45:49.694688-0600 Loop[432:152291] [PeripheralManager+RileyLink] RL Send: 19050000000000000000b4030000a966726b18f468d555359000
2018-07-20 12:45:50.477870-0600 Loop[432:152687] [PeripheralManager+RileyLink] RL Recv(single): dd6b51a966726b18f468d5595638e58f28e3555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555c8d500
[... cut ...]
2018-07-20 12:51:59.766814-0600 Loop[432:154656] [PeripheralManager+RileyLink] RL Recv(single): dd6684a966726bdd6785a96672dd6786a96672dd6787a96672aa
2018-07-20 12:51:59.767138-0600 Loop[432:154656] [PeripheralManager+RileyLink] RileyLink response: PacketResponse(code: RileyLinkBLEKit.ResponseCode.success, packet: Optional(RileyLinkBLEKit.RFPacket(data: 23 bytes, packetCounter: 132, rssi: -22)))
2018-07-20 12:51:59.767862-0600 Loop[432:154164] [RileyLinkDevice] ------------------------ Pump ID Setup ---------------------------
2018-07-20 12:51:59.769036-0600 Loop[432:154164] [RileyLinkDevice] Session queue operation count is now empty

dm61, 2018-07-23:
An update: I've been able to complete the pump setup after commenting out `try session.selectBasalProfile(.standard)` line in `MinimedPumpIDSetupViewController.swift`. 

Looks like there is something odd in the way my 523 responds to that particular command, @ps2 let me know if I could do anything else to help in debugging this issue. Has anyone else tried the latest Loop dev with an x23 pump? 

BTW, why do we even need that command in the pump setup?

ps2, 2018-07-24:
We have paired other x23 pumps, so I'm not sure what's going on here.  Haven't had time to dig into it yet.

dm61, 2018-07-24:
Good to hear you've paired other x23 pumps. So, it's a more special case, maybe particular pump firmware, mine is `VER 2.3A 1.1 0B 0B`, and RL is `subg_rfspy 2.0/ble_rfspy 2.0`  In any case, I am fine with a temporary workaround, nothing urgent.

Kdisimone, 2018-07-24:
I’ll try my x23 pumps and see if I can recreate

Kdisimone, 2018-07-24:
@dm61 just curious, did it matter whether you had patterns enabled or disabled in the pump when you were trying to add the pump?  Did you try both?

alternateal, 2018-07-24:
Or go back to v1.56 Master which doesn’t perform this check up front

alternateal, 2018-07-24:
@Kdisimone - worked perfectly with basal patterns disabled on 1.5.7 dev

itsmojo, 2018-07-24:
Last night I tried a running Loop dev on my 723 running VER 2.4A 1.1 0B 0B and was getting “Unknown response during PumpMessage (Carelink, selectBasalProfile, 3 bytes, 65 bytes):24 bytes” during Add Pump->RileyLink Setup. Just tried this morning and got the same error, but with 22 bytes instead of the 24 bytes I saw last night and the 23 bytes that @alternateal saw. I was also getting thrown into Xcode in RileyLinkDeviceTableViewCell.setup() with an EXEC_BAD_ACCESS error on the “let connectSwitch =“ line. Seems like I too have a troublesome pump. Let me know if there is anything that I can do to assist with this one.

[image]
[image]
[image]

Kdisimone, 2018-07-24:
@itsmojo I think just turning off patterns in the basal menu on pump should fix it

Kdisimone, 2018-07-24:
All the write commands go to the standard profile, I remember seeing when I was looking at the commits...that’s what made me think patterns may be playing a part. Nothing is mapped for pattern a or pattern b

dm61, 2018-07-27:
I've just built and tested Loop with the rileylink_ios fix #421, and the setup went through without any problems. Thanks @ps2!

PR #421 (2018-07-26): Handle slower pump response to set profile command, and speedup tune
Fixes #420
closed 2018-08-25 by dm61