ps2/rileylink_ios #453: OmniKit: Deactivate Pod command is missing an alert settings reset

Lytrix, 2018-09-09:
The Deactivate Pod command must also receive the 
`Command 02 Status Request type $46 - flash variables including state, initialization time, any faults`
Else the Pod will continue to issue expiration beeps even after deactivating.

ps2, 2018-09-25:
Could you post an example of the command that is set during deactivation?

Lytrix, 2018-09-26:
I also do not see any $19 command, this is probably set more specifically at the priming stage. 
But I want to test ending a pod by also sending the $46 $50 and $51 commands, to see if that does anything.

ps2, 2018-09-26:
I think this ticket is probably more an artifact of the fact that the changePod() call wasn't being made at all, when the user tapped Deactivate Pod in the UI.  It is now: https://github.com/ps2/rileylink_ios/blob/dev/OmniKitUI/OmnipodSettingsViewController.swift#L319

Lytrix, 2018-09-27:
Ah yes, there was indeed no changePod() action called. Great, will test this soon when I start recording all the different pod info types probably this weekend.

Lytrix, 2018-10-11:
Deactivation now works properly with the added fault event  fallback, thanks!

closed 2018-10-11 by Lytrix