ps2/rileylink_ios #466: Is resync Nonce issue fixed?

dexdan, 2018-09-23:
RileyLink (Dev branch)
Dan Evans 23/9/18

The attempt to send a second bolus via the test command button in the app says ’succeeded’ but there is no actual bolus delivered. 

Slack post by @lytrix 23/9/189 1:10 suggests this may be a resync nonce issue - but it doesn’t seem to be fixed for me?

This is the iPhone console from when the second test (bolus) was attempted. 

(Note that issuing a second bolus in this way DID work when the 3 lines including and following OmniKit/PumpManager/PodCommsSession.swift:244 and 266 are commented out - as @lytrix says).

default	23:29:59.271746 +0100	assertiond	Updating PowerAssertion on RileyLink:345
default	23:29:59.272629 +0100	assertiond	Client relinquished <BKProcessAssertion: 0x137d3b9f0; "com.apple.UIKit.KeyboardManagement.message" (finishTask:180s); id:…AB1067CC0AD2>
default	23:29:59.277466 +0100	assertiond	[RileyLink:345] Deactivate assertion: <BKProcessAssertion: 0x137d3b9f0; "com.apple.UIKit.KeyboardManagement.message" (finishTask:180s); id:…AB1067CC0AD2>
default	23:29:59.277923 +0100	assertiond	[RileyLink:345] Setting jetsam priority to 10 [0x10110]
default	23:29:59.279335 +0100	assertiond	Updating PowerAssertion on RileyLink:345
default	23:29:59.283534 +0100	assertiond	[RileyLink:345] Remove assertion: <BKProcessAssertion: 0x137d3b9f0; "com.apple.UIKit.KeyboardManagement.message" (finishTask:180s); id:…AB1067CC0AD2>
default	23:29:59.284459 +0100	assertiond	Updating PowerAssertion on RileyLink:345
default	23:29:59.341718 +0100	assertiond	-[BKAssertion dealloc] - <0x137d3b9f0>
default	23:29:59.350443 +0100	backboardd	[HID] [MT] dispatchEvent Dispatching event with 1 children, _eventMask=0x2 _childEventMask=0x2 Cancel=0 Touching=0 inRange=1
default	23:29:59.789793 +0100	CommCenter	QMI: Svc=0x03(NAS) Ind MsgId=0x0051 Bin=['01 15 00 80 03 02 04 00 00 51 00 09 00 14 06 00 BB F9 A3 FF 94 00']
default	23:30:00.037036 +0100	powerlogHelperd	{"msg":"CLCopyAppsUsingLocation", "event":"activity"}
default	23:30:00.395887 +0100	symptomsd	Collect info, last update was -0.000007 secs ago
default	23:30:00.398134 +0100	symptomsd	CFSM polled info succinct report: Negative
default	23:30:00.536923 +0100	locationd	Current state of Workout Predictor: 0
default	23:30:01.032787 +0100	powerlogHelperd	{"msg":"CLCopyAppsUsingLocation", "event":"activity"}
default	23:30:01.859235 +0100	symptomsd	L2 Metrics on ifname en0: rssi: -82 (txFrames/txReTx/txFail) 1/1/0 -> (was/is) 0/0
default	23:30:01.898066 +0100	symptomsd	Data Stall count: 0
default	23:30:01.898240 +0100	symptomsd	Data Stall score: 50
default	23:30:01.898502 +0100	symptomsd	Policy denial score: 0.000000
default	23:30:01.898692 +0100	symptomsd	More info Foreground app <present>  interval 0.000000 flows 0 (total 5)
default	23:30:01.898851 +0100	symptomsd	Also flags 1  count 0  details Foreground app <present>  interval 0.000000 flows 0 (total 5)
default	23:30:01.902498 +0100	wifid	WiFi:[559434601.901365]: __WiFiLQAMgrCheckTriggerDisconnect_block_invoke Symptoms Input: {
    "AWD-info" =     {
        "ARP-failures" = 0;
        "data-stalls-score" = 50;
        "dns-impacted" = 0;
        "dns-total" = 1;
        "gate-open-duration" = 0;
        "rnf-denials-score" = 0;
        "valid-state" = 1;
        "watch-duration" = 0;
        "watch-rx-count" = 0;
        "watch-time-since-now" = 0;
    };
    impact = 0;
    "impact-elapsed-since-close" = 0;
    "impact-flags" = 1;
    "impact-reasons" = "Foreground app <present>  interval 0.000000 flows 0 (total 5) ";
    "net-score" = 50;
    "net-score-reasons" =     {
        "ARP-state" =         {
            score = 50;
            "score-info" = "Current ARP symptoms: 0";
        };
        "DNS-state" =         {
            score = 50;
            "score-info" = "Current DNS servers total 1, impacted : 0 valid 1";
        };
        "Flow-state" = "No further info";
        "ND6-state" =         {
            score = 50;
            "score-info" =
default	23:30:01.903074 +0100	wifid	WiFi:[559434601.902560]: __WiFiLQAMgrSymptomsInputCallBack Proceeding with Trigger Discconect - DataStallScore 50 AppPolicyScore 0.000000
default	23:30:01.903249 +0100	wifid	WiFi:[559434601.902812]: __WiFiLQAMgrSymptomsInputCallBack Foreground Networking App not present. Overriding TrgDisc decision
default	23:30:01.903676 +0100	wifid	WiFi:[559434601.903039]: __WiFiLQAMgrSymptomsInputCallBack: TrgDiscStatus 36872 [Rssi -82 TxFrms 1 TxFail 0 TxRet 1 fwFrms 2 fwFail 0 fwRtry 1 RxFrms 1 BcnSch 95 BcnRx 8 Awd 0 UnGw 0.000000 FgApp 1 Walkout 0 possibleRoam 1 DataStallScore 50 DeltaTime -1.000000secs TD1 0]
default	23:30:02.019941 +0100	powerlogHelperd	{"msg":"CLCopyAppsUsingLocation", "event":"activity"}
default	23:30:02.193900 +0100	assertiond	Client relinquished <BKProcessAssertion: 0x137d4a0d0; "com.rileylink.rlapp" (bluetooth:inf); id:…90DF9D158F80>
default	23:30:02.197794 +0100	assertiond	[RileyLink:345] Deactivate assertion: <BKProcessAssertion: 0x137d4a0d0; "com.rileylink.rlapp" (bluetooth:inf); id:…90DF9D158F80>
default	23:30:02.198287 +0100	assertiond	[RileyLink:345] Scheduling allow-idle-sleep timer with interval: 180.0
[... 546 lines cut ...]
    };
    impact = 0;
    "impact-elapsed-since-close" = 0;
    "impact-flags" = 1;
    "impact-reasons" = "Foreground app <present>  interval 0.000000 flows 0 (total 5) ";
    "net-score" = 50;
    "net-score-reasons" =     {
        "ARP-state" =         {
            score = 50;
            "score-info" = "Current ARP symptoms: 0";
        };
        "DNS-state" =         {
            score = 50;
            "score-info" = "Current DNS servers total 1, impacted : 0 valid 1";
        };
        "Flow-state" = "No further info";
        "ND6-state" =         {
            score = 50;
            "score-info" =
default	23:31:02.751258 +0100	wifid	WiFi:[559434662.749704]: __WiFiLQAMgrSymptomsInputCallBack Proceeding with Trigger Discconect - DataStallScore 50 AppPolicyScore 0.000000
default	23:31:02.751463 +0100	wifid	WiFi:[559434662.749926]: __WiFiLQAMgrSymptomsInputCallBack Foreground Networking App not present. Overriding TrgDisc decision
default	23:31:02.751678 +0100	wifid	WiFi:[559434662.750288]: __WiFiLQAMgrSymptomsInputCallBack: TrgDiscStatus 36872 [Rssi -84 TxFrms 0 TxFail 0 TxRet 0 fwFrms 16 fwFail 0 fwRtry 4 RxFrms 1 BcnSch 95 BcnRx 6 Awd 0 UnGw 0.000000 FgApp 1 Walkout 0 possibleRoam 1 DataStallScore 50 DeltaTime -1.000000secs TD1 0]
default	23:31:03.028806 +0100	powerlogHelperd	{"msg":"CLCopyAppsUsingLocation", "event":"activity"}
default	23:31:03.770861 +0100	CommCenter	QMI: Svc=0x03(NAS) Ind MsgId=0x0051 Bin=['01 15 00 80 03 02 04 00 00 51 00 09 00 14 06 00 BC F5 A1 FF 14 00']
default	23:31:04.028239 +0100	powerlogHelperd	{"msg":"CLCopyAppsUsingLocation", "event":"activity"}

Lytrix, 2018-09-28:
@dexdan could you test this branch if it is solved here? We are going to use this branch for now to develop some large parts separate from the dev branch. https://github.com/ps2/rileylink_ios/tree/omnipod-testing

ps2, 2018-09-28:
The failure to bolus on second try is not a nonce resync issue; it's just an artifact of running in the RL app, which doesn't finalize doses.  If you change the testingCommands() function to look like this, I think it will work:
```
    public func testingCommands() throws {
        let response: StatusResponse = try send([GetStatusCommand()])
        finalizeDoses(deliveryStatus: response.deliveryStatus) { (_) -> Bool in
            return true
        }
        //try setTempBasal(rate: 2.5, duration: .minutes(30), confidenceReminder: false, programReminderInterval: .minutes(0))
        //try cancelDelivery(deliveryType: .tempBasal, beepType: .noBeep)
        let _ = bolus(units: 20)
        //try cancelDelivery(deliveryType: .bolus, beepType: .bipBip)
    }
```

ps2, 2018-09-28:
I just pushed an update to the omnipod-testing branch that has this and also reports errors from the bolus command.

closed 2018-09-28 by ps2

dexdan, 2018-09-29:
> The failure to bolus on second try is not a nonce resync issue; it's just an artifact of running in the RL app, which doesn't finalize doses. If you change the testingCommands() function to look [...]

Thanks @ps2. Having a spot of bother with build errors on omnipod-testing branch - have asked for help in slack group and will retest once resolved. Many thx.