ps2/rileylink_ios #497: Unexpectedly found nil while unwrapping an Optional value

dthornley, 2019-03-03:
When using the omnipod-testing branch the above error is being shown when I run RileyLink on my iPhone via xcode. The error is being shown when I click on cancel or pair in the Pod Pairing screen.

The errors are being raised from the functions cancelButtonPressed() and pair() in PairPodSetupViewController.swift when the pumpManager is being used.

I'm new to swift, but I did notice that pumpManager is still nil when the getPodState or pairAndPrime methods are called.

matthewjweaver, 2019-03-04:
I can reproduce reliably by clicking cancel after arriving on the pod setup dialog.

dthornley, 2019-03-07:
Logs from Xcode when clicking on Cancel:

`2019-03-07 18:58:44.447840+0100 RileyLink[14016:2864356] [RileyLinkDeviceManager] centralManagerDidUpdateState: Powered On
2019-03-07 18:58:44.490171+0100 RileyLink[14016:2864356] [RileyLinkDeviceManager] Discovered <CBPeripheral: 0x281649040, identifier = DF339946-4D62-2EBB-4F4E-65BC7A259AC6, name = RileyLink, state = disconnected> at -66
2019-03-07 18:58:44.509875+0100 RileyLink[14016:2864355] [PeripheralManager] Error applying peripheral configuration: notReady
2019-03-07 18:58:44.509934+0100 RileyLink[14016:2864355] [RileyLinkDevice] Reading firmware versions for PeripheralManager configuration
2019-03-07 18:58:44.511522+0100 RileyLink[14016:2864355] [PeripheralManager] Error applying delegate configuration: peripheralManagerError(RileyLinkBLEKit.PeripheralManagerError.unknownCharacteristic)
2019-03-07 18:58:44.516396+0100 RileyLink[14016:2864356] [RileyLinkDeviceManager] Created device for peripheral <CBPeripheral: 0x281649040, identifier = DF339946-4D62-2EBB-4F4E-65BC7A259AC6, name = RileyLink, state = disconnected>
2019-03-07 18:58:44.516758+0100 RileyLink[14016:2864358] [PeripheralManager] Error applying peripheral configuration: notReady
2019-03-07 18:58:44.516999+0100 RileyLink[14016:2864358] [RileyLinkDevice] Reading firmware versions for PeripheralManager configuration
2019-03-07 18:58:44.519205+0100 RileyLink[14016:2864358] [PeripheralManager] Error applying delegate configuration: peripheralManagerError(RileyLinkBLEKit.PeripheralManagerError.unknownCharacteristic)
2019-03-07 18:58:44.545823+0100 RileyLink[14016:2864243] applicationDidBecomeActive
2019-03-07 18:58:45.139787+0100 RileyLink[14016:2864358] [RileyLinkDeviceManager] Discovered <CBPeripheral: 0x281649040, identifier = DF339946-4D62-2EBB-4F4E-65BC7A259AC6, name = RileyLink, state = connecting> at -65
2019-03-07 18:58:45.168726+0100 RileyLink[14016:2864358] [RileyLinkDevice] didConnect <CBPeripheral: 0x281649040, identifier = DF339946-4D62-2EBB-4F4E-65BC7A259AC6, name = RileyLink, state = connected>
2019-03-07 18:58:45.220909+0100 RileyLink[14016:2864356] [RileyLinkDevice] didConnect <CBPeripheral: 0x281649040, identifier = DF339946-4D62-2EBB-4F4E-65BC7A259AC6, name = RileyLink, state = connected>
2019-03-07 18:58:45.315415+0100 RileyLink[14016:2864355] [PeripheralManager] Peripheral configuration completed
2019-03-07 18:58:45.315462+0100 RileyLink[14016:2864355] [RileyLinkDevice] Reading firmware versions for PeripheralManager configuration
2019-03-07 18:58:45.474961+0100 RileyLink[14016:2864355] [PeripheralManager+RileyLink] RL Send (single): 0102
2019-03-07 18:58:45.585893+0100 RileyLink[14016:2864358] [PeripheralManager+RileyLink] RL Recv(single): dd737562675f726673707920322e32
2019-03-07 18:58:45.587905+0100 RileyLink[14016:2864358] [PeripheralManager+RileyLink] RileyLink response1: GetVersionResponse(code: RileyLinkBLEKit.ResponseCode.success, version: "subg_rfspy 2.2")
2019-03-07 18:58:45.588085+0100 RileyLink[14016:2864355] [PeripheralManager] Delegate configuration completed`

Logs for when you click on Pair are the same.

dthornley, 2019-03-07:
Screenshot of the UI for the above logs.

[image]

Lytrix, 2019-03-09:
I've took a shot at looking into this. It looks like nil is returned on the OmnipodPumpManager which is Force unwrapped here:
https://github.com/ps2/rileylink_ios/blob/omnipod-testing/OmniKitUI/ViewControllers/PairPodSetupViewController.swift#L19
It returns a nil here:
[image]

Could it be that the OmnipodPumpManager cannot be loaded into the pumpManager variable? @ps2 do you have any pointers or an example if I am on the right track or where I could look further or maybe a similar example of the Medtronic loading of the pumpmanager?

ps2, 2019-03-09:
The hosting nav controller should be passing the pump manager to its viewcontrollers.

jasoncalabrese, 2019-03-16:
Is there a fix for this?

Lytrix, 2019-03-31:
@jasoncalabrese Not yet, The nil is returned because the omnipod pairing storyboard in xcode has a page before the pairing steps are starting to setup a basal schedule. In our Loop test setup this route is correctly shown the first time setting it up, but in rileylink_ios this page is apparently ignored on the first run resulting in a nil basal schedule. I do not have worked enough with storyboards yet to know how to reroute this in a proper way.