ps2/rileylink_ios #516: Pod state not propagated to NS

panctronic, 2019-04-12:
It seems like complete (reservoir/battery at least) Pod state isn't making it to the Nightscout pump pill.

I'm currently getting records like:
`   "pump": {
        "clock": "2019-04-12T00:17:41Z",
        "suspended": false,
        "pumpID": "XXXXXXXX",
        "bolusing": true,
        "secondsFromGMT": -25200
    },
`
Corresponding to a pill like:
[image]

Compared to a typical minimed sourced entry:
`
    "pump": {
        "clock": "2019-02-21T06:01:00Z",
        "reservoir": 249.6,
        "secondsFromGMT": -28800,
        "battery": {
            "percent": 100
        },
        "pumpID": "000000"
    }
`

Since pods don't report volume over 50u or battery level, resolving this issue may require changes in several places to support the additional logical state (pod lifetime remaining, etc)

jasoncalabrese, 2019-04-24:
I looked at this the other day and was wondering why `bolusing` always seems to be true

panctronic, 2019-05-09:
https://github.com/nightscout/cgm-remote-monitor/pull/4578 will add 50+ Unit display in NS and depends on https://github.com/LoopKit/Loop/pull/947 and https://github.com/ps2/rileylink_ios/pull/522

francescaneri, 2019-05-09:
Sensor updates at 7 minutes, pump and loop at 5 minutes.  I don't see pill (50+ Unit)

jasoncalabrese, 2019-05-09:
If you’ve already updated the loop app and your NS site it might be cached by your browser, try doing some hard refreshes.