ps2/rileylink_ios #558: Suspending a Pod during a temp basal confuses Loop

itsmojo, 2019-10-11:
Taken from https://loop.zulipchat.com/#narrow/stream/144120-omnipod_dev/topic/Suspend.20during.20temp.20basal/near/177870577

With the current Loop dev if you tap on `Suspend Delivery` while a temp basal is currently in effect, Loop gets into a funky internal state where the Pod is suspended but Loop doesn’t think it is. The user will see that the button will transition to a busy state and then return back to `Suspend Delivery` as if the suspend didn’t happen, but the Pod is actually suspended and is not delivering any insulin (i.e., the Pod cancel all command was successful). In this bogus state when the Pod is actually suspended the home screen shows a basal rate of +0.0 U and does not say “__Pump Suspended__   Tap to Resume” while the Pod Settings screen shows “Basal Delivery” of “Schedule” as well as having the action button which says `Suspend Delivery`. When in this confused state, new temp basals cannot be started however because the actual Pod status is being used and the operation is cancelled (Xcode logs say “Canceling temp basal because status return indicates pod is suspended.”), but Loop doesn’t update its internal state with this status so the confused & inconsistent Loop state remains. Eventually the HUD loop will turn to yellow and then red if nothing else is done and this is the only indication that there is any problem (without even giving any indication that there is no insulin delivery).

The user recovery from this bogus state is either to tap `Suspend Delivery` again and afterwards Loop will actually believe that the Pod is suspended (because no TB was in effect when the suspend was done) and will now display `Resume Delivery`. Then just tap on `Resume Delivery` and everything will be running properly again. Another alternate recovery method from this bogus state is to deliver any non-zero bolus which has a Loop side effect of resuming the suspended Pod.

I verified that the return from the cancel all Pod command when doing the `Suspend Delivery` when a temp basal is currently active does returns a normal suspended state, but then something must be going wrong during the insulin dose accounting work for the canceled temp basal in this situation that leaves the Loop's internal state believing that the Pod is running on the normally scheduled basal rate instead of being suspended (i.e., like a cancel temp basal had occurred instead of a cancel all).

closed 2019-12-03 by itsmojo

loyalchallenger, 2019-12-25:
I have experienced this also. It would be nice if there were a way to cancel a temp basal in such a way that it simply reverts back to your scheduled basal. Otherwise, I have to click Suspend Delivery twice and then Resume Delivery.

itsmojo, 2019-12-27:
The original suspend problem when a temp basal is active with a Pod was fixed with https://github.com/ps2/rileylink_ios/pull/568, so now a single click on Suspend Delivery should always suspend the Pod (independent of whether a temp basal is in effect or not) and then a Resume Delivery would then take you back to the scheduled basal.

@loyalchallenger This is a request for new functionality for all pump types and really shouldn't have been included in this particular Issue Report, but I'd say that it is probably not worth adding a new separate way to cancel just a temp basal for Loop since two consecutive presses on the same button will get you there and this shouldn't be that common of an operation.

loyalchallenger, 2019-12-27:
My point was simply that it is not displaying that it is actually suspended. It just shows that the temp basal was cancelled, but the pod is suspended and then I have to click suspend delivery again so that I can click resume delivery. For new users, it should be made more clear.

itsmojo, 2019-12-27:
@loyalchallenger You are now describing the problem that this Issue Report was originally filled against which has already been fixed. Build a new copy of Loop dev and you should no longer see the buggy & incorrect behavior you are describing. Suspend Delivery suspends all insulin delivery and now appropriately updates all its internal state to reflect this in a single tap.