ps2/rileylink_ios #569: Pod non badNonce Error Responses are not decoded, logged or treated as a certain failure

itsmojo, 2019-12-01:
If a Pod returns an 06 Error Response for anything but 0x14 (badNonce) error, rileylink_ios fails to decode the response which results the message not being added to the MessageLog and an uncertain failure. These error responses should be properly decoded, logged in the MessageLog, and treated as a certain failure.

itsmojo, 2019-12-02:
PR #571 fixes one of the 3 problems reported in this issue report. By moving the one line in PR #571, Loop will now log all message responses in both the MessageLog and in Xcode debug output even if the message cannot be successfully parsed (i.e, it fixes the not being logged problem). This fix is not specific to non badNonce Error Responses and should land independently ASAP.

Additional more complicated code fixes are still needed to fix the other 2 problems of non badNonce Error Responses not being decoded and not being treated as a certain failure.

ps2, 2019-12-07:
In what cases does the pod issue an 06 with a non 0x14 type? Is that something that happens normally?

itsmojo, 2019-12-08:
It would be difficult to enumerate the various possible error responses, but they aren’t normal. There are maybe a hundred different firmware routines that return a non-zero error values (<= 20) to their caller. Depending on the situation, the caller could either ignore the error return, fault the Pod, or the return that error (or some other error) to a higher level routine and/or back to the PDM with an error status return. The reproducible error that I had involved a message formatted with an additional command along with an insulin command which the Pod unfortunately disallows with an 07 error response.  There are a number of different errors for various command/message formatting issues like this, but there are also potential transient errors that could also be returned. In any case, all these error returns should be able to be properly decoded and treated as a certain failure for things like setting temp basals.

ps2, 2019-12-08:
Sounds like we should track a `primeUnitsDelivered` variable in pod state.

itsmojo, 2020-04-15:
When merged, https://github.com/ps2/rileylink_ios/pull/596 will decode and return a certain error from send() for any non badNonce error response and this issue can be closed. Tested by sending an additional sub-message with a bolus command which will generate an error response with code 7.

itsmojo, 2021-03-01:
Fixed Sep 30, 2020 with merge of #596

closed 2021-03-01 by itsmojo