ps2/rileylink_ios #682: Connection to EmaLink can be intermittent on dev branch

marionbarker, 2021-09-20:
EmaLink could not connect without power cycle of phone and EmaLink.
This may be similar to an issue fixed with v2.2.5 version of the code.
Zulipchat report:
* https://loop.zulipchat.com/#narrow/stream/144182-development/topic/dev.3A.20Ema.20question/near/253983158

Copy of that zulipchat message for completeness:

I captured an Xcode log for a case where both real EmaLink and a RileyLink running EmaLink firmware had a problem with dev code. Stopped being able to communicate with pump with either link using my "real" Loop dev phone.

Tried without success:
* power cycle link
* close and restart Loop

Specific devices having problems (both devices show subg and ble on backup phone running v2.2.6):
* A: RileyLink hardware with EmaLink firmware (subg_rfspy 2.2.19 / ble_rfspy 3.11) - showing only ble with dev
* B: EmaLink (subg_rfspy 2.2 / ble_rfspy 3.17) - showing only ble with dev

Steps:
* I built Loop dev (no patches for rileylink_ios) onto "real" phone at 2021-09-19 16:59 so I could capture the log.
* Both A and B were showing the ble_rfspy only before I built Loop dev on top of existing Loop dev app.
* I touched A in pod settings menu: and it then reported both subg as well as ble firmware. (still 16:59)
* I touched B in pod settings menu: and it remained unable to communicate. (17:00)
* Tried power cycle of B and still only ble showed.
* Disconnected from Xcode.  Power cycled phone.  Had to also power cycle B before I could connect.  

[20210919_1659_to_1703_dev_Xcode_debug_log_EL.txt](https://github.com/ps2/rileylink_ios/files/7193197/20210919_1659_to_1703_dev_Xcode_debug_log_EL.txt)

marionbarker, 2021-10-01:
I looked at the modifications that were implemented to fix the "ema" problem and to clean up some "orange" comms leading up to loop-release/v2.2.5 (commit 15d1997 of rileylink_ios).

I believe the attached patch captures the modifications needed for dev branch to have these same improvements.

I have been using this patch successfully on dev branch on myself using EmaLink and OrangeLink Pro for more than one week. The naming of the patch might need some explanation.  The "master" referred to in the name is the LoopWorkspace build for master that points to rileylink_ios commit 15d1997.

I started with LoopWorkspace build of dev branch, which points to commit 44eab01 of rileylink_ios:
```
cd rileylink_ios
git apply dev_RL_BLE_UI_44eab01_to_master_15d1997.txt
cd ..
xed .
```

This patch modifies the following files, and includes the Xcode 13 plus Xcode 12 code changes:
* RileyLinkBLEKit/PeripheralManager+RileyLink.swift
* RileyLinkBLEKit/PeripheralManager.swift
* RileyLinkBLEKit/PeripheralManagerError.swift
* RileyLinkBLEKit/Response.swift
* RileyLinkBLEKit/RileyLinkDevice.swift
* RileyLinkKitUI/RileyLinkDeviceTableViewController.swift

[dev_RL_BLE_UI_44eab01_to_master_15d1997.txt](https://github.com/ps2/rileylink_ios/files/7267380/dev_RL_BLE_UI_44eab01_to_master_15d1997.txt)

marionbarker, 2021-10-03:
I realized the name is not actually very good.  (That symptom is what I noticed first.) So changed the name.

rolandet, 2022-01-22:
@marionbarker Here are the logs and export data reports from an Pod Pairing issue that I experienced with your patch dev_RL_BLE_UI_44eab01_to_master_15d1997.txt patch.  Unfortunately the data was collected after I Paired another Pod successfully right afterward so I do not know how helpful this is. As I mentioned on Zulip, I've only seen two Pod Pairing issues with your patch in the 7 weeks. Do you have any plans to submit a PR request for this patch to the dev branch?

[Export-20211225T224651Z.zip](https://github.com/ps2/rileylink_ios/files/7917523/Export-20211225T224651Z.zip)
[Loop Report 2021-12-25 17_46_14-05_00.md](https://github.com/ps2/rileylink_ios/files/7917524/Loop.Report.2021-12-25.17_46_14-05_00.md)

rolandet, 2022-01-22:
@marionbarker Good news. I had another Pod failure today and was able to capture the Issue Report and  the Critical Events Export. I was also able to capture a couple of screen captures. All of these are attached below.  The short of it, the error was 092 prime count too low.

092 | 0x5C | Prime open count too low | Problem with a safety check after priming

[image]
[image]
[Export-20220122T165526Z.zip](https://github.com/ps2/rileylink_ios/files/7918935/Export-20220122T165526Z.zip)
[Loop Report 2022-01-22 11_52_53-05_00.md](https://github.com/ps2/rileylink_ios/files/7918936/Loop.Report.2022-01-22.11_52_53-05_00.md)

marionbarker, 2022-07-05:
dev branch updated a while ago.  Closing this.

closed 2022-07-05 by marionbarker