ps2/rileylink_ios #686: Updates Needed for Ema and Orange Displays

marionbarker, 2021-10-01:
Updates are needed to handle Ema and Orange Displays for the LoopWorkspace build for master that points to rileylink_ios commit 15d1997.

The items displayed in the RileyLink Display need some updates to handle:
* EmaLink
  * Battery display
  * "Invert LED Logic" (actual function of "Diagnostic LEDs" row for .riley)
* OrangeLink and OrangeLink Pro
  * Interpret hardware type from all released versions of OL and OL Pro firmware
  * Show the "Find Device" line for HW1.1 (OL Pro) regardless of firmware

The enclosed patch has been used in real life for more than a week. It has been tested with OL and OL Pro with every firmware version up through FW3.2, as well as for EmaLink. It is based off suggestions from Eric Jensen on zulipchat.

The patch can be applied to the LoopWorkspace build for master that points to rileylink_ios commit 15d1997:
```
cd rileylink_ios
git apply ema_ol_fix_master_15d1997.txt
cd ..
xed .
```

This patch modifies the following 2 files:
* RileyLinkBLEKit/RileyLinkDevice.swift
* RileyLinkKitUI/RileyLinkDeviceTableViewController.swift

[ema_ol_fix_master_15d1997.txt](https://github.com/ps2/rileylink_ios/files/7267821/ema_ol_fix_master_15d1997.txt)

jlucasvt, 2021-10-04:
Marion. I'm running the latest L&L build script workspace and get an error when trying to apply the the patch.

marionbarker, 2021-10-05:
Screenshot?

jlucasvt, 2021-10-05:
Yes I have XCode 13 and successfully build and deplyed to phone and watch.  I copied the patch to my rileylink_ios folder and attempted apply.

[image]

marionbarker, 2021-10-05:
You had already made changes to those 2 files.  You can only apply a patch to a fresh copy of rileylink_ios HEAD detached at 15d1997. (Or at least those two files cannot be modified from fresh copy.)

Try these commands to fix what you currently have. 
This moves the patch out of directory up one level, otherwise stash will remove it.

mv ema_ol_fix_master_15d1997.txt ..

Restore clean copy of rileylink_ios

git stash

Now apply the patch 

git apply ../ema_ol_fix_master_15d1997.txt

jlucasvt, 2021-10-10:
Working great on my OL Pro.  Will do some further testing this week no problems thus far.

sumairs, 2022-01-05:
Hi @marionbarker, I got a bunch of build errors when applying the patch to loop V2.2.6. I have an OL Pro FW3.2/HW1.1. 
Any idea what I may have done wrong?

[image]
[image]

ps2, 2022-01-17:
Are all of these changes in dev at this point?  If not, please identify what is remaining. If a patch can be created, I can review it, or I can look into getting the remaining issues resolved.

marionbarker, 2022-01-17:
None of these changes are in dev yet.
Using this commit: 6321592:
* OrangeLink Pro does not display Find Device
* EmaLink does not display battery, battery level alert or Invert LED Logic row

I did 
```
git apply ema_ol_fix_master_15d1997_vs2.txt
```

and displays of both are as desired.

closed 2022-01-18 by ps2

marionbarker, 2022-07-05:
@ps2 I just looked at this again today based off a facebook comment.  I think there was some confusion when you closed this issue.  It's fixed in dev but not in master.  I've updated the patch instructions to include the updated commit number for rileylink_ios.  

We point to the [patch instructions in LoopDocs](https://loopkit.github.io/loopdocs/operation/loop-settings/rileylink/#rileylink-status-and-commands). Scroll down to the admonition that says `Patch Required for Loop 2.2.x`